rust-embedded/cargo-binutils #6: cargo-size UI

japaric, 2018-06-29:
This issue is about improving the UI of the `cargo-size` subcommand to make it feel more integrated
with Cargo. Possible additions:

- `cargo size [--target $T] --example $E [--release] -- $ARGS`. Builds `$E` for `$T` and then calls
  `llvm-size $ARGS target/$T/{debug,release}/examples/$E`.

- `cargo size [--target $T] --examples [--release] -- $ARGS`. Builds *all* the examples and then
  calls `llvm-size $ARGS target/$T/{debug,release}/examples/*`.

Thoughts? Suggestions?

pravic, 2018-06-29:
>   Print size in radix:
    -o                - Print size in octal
    -d                - Print size in decimal
    -x                - Print size in hexadecimal

Addresses should be always in hexadecimal form.

Right now it looks awkward:

```
section       size         addr
.text       171656   5368713216
.rdata       49016   5368885248
.data          512   5368934400
.pdata       11340   5368938496
.reloc         728   5368950784
Total       233252
```

burrbull, 2018-07-16:
Do not forget about `target-dir`.
This path can be changed in `.cargo/config`

japaric, 2018-08-28:
v0.1.2 implements `--bin` `--example`, `--lib`, `--release` and `--target` flags which can be used to build a binary / example / library *and* inspect the build artifact -- no need to pass the path to the artifact as an argument. [Examples](https://github.com/rust-embedded/cargo-binutils/blob/v0.1.2/README.md#size).

if there's any other UI improvement you'd like to see please open a new issue. I'm closing this one as "done".

closed 2018-08-28 by japaric

japaric, 2018-08-28:
> Do not forget about target-dir.

Yeah, I totally forgot. :sweat_smile:

I have opened a help wanted issue w/ instructions if you fill like fixing this! #24