rust-embedded/cargo-binutils #7: error: llvm-tools component is missing or empty

pravic, 2018-06-29:
> $ rustup component add llvm-tools --toolchain nightly
> $ cargo +nightly install cargo-binutils

> $ cargo +nightly size
error: `llvm-tools` component is missing or empty. Install it with `rustup component add llvm-tools`

> $ rustup component list --toolchain nightly | grep tools
llvm-tools-x86_64-pc-windows-msvc (installed)

Version: 
> $ rustc +nightly -vV
rustc 1.28.0-nightly (cd494c1f0 2018-06-27)
binary: rustc
commit-hash: cd494c1f0915da00a63c03454a96d504afe764ff
commit-date: 2018-06-27
host: x86_64-pc-windows-msvc
release: 1.28.0-nightly
LLVM version: 6.0

japaric, 2018-06-29:
@pravic can you check the directory `$(rustc --print sysroot)/bin` and see if `llvm-size` is in there?

My guess is that it is there but it's named `llvm-size.exe`. We'll need to update the search algorithm to account for this.

pravic, 2018-06-30:
Yes, those tools are there and the `rustup run nightly llvm-size` does work.

japaric, 2018-06-30:
> rustup run nightly llvm-size does work.

That's not intended to work and will stop working when rust-lang/rust#51922 lands.

@pravic Could you try out #8?

japaric, 2018-07-15:
Fixed in #8

closed 2018-07-15 by japaric