rust-embedded/cargo-binutils #12: error: component 'llvm-tools' is unavailable for download

pravic, 2018-07-13:
I don't know where is a bug tracker for the llvm-tools, so let me place it here.

> info: syncing channel updates for 'nightly-x86_64-pc-windows-msvc'
info: latest update on 2018-07-13, rust version 1.29.0-nightly (64f7de921 2018-07-12)
error: component 'llvm-tools' for 'x86_64-pc-windows-msvc' is unavailable for download

It's not about the "llvm-tools-preview" (NOTE The rustup component might be renamed to llvm-tools-preview to better reflect the lack of stability guarantees.), the component is completely missing:

> $ rustup component list --toolchain nightly | grep llvm
>
> $

japaric, 2018-07-13:
> I don't know where is a bug tracker for the llvm-tools

rust-lang/rust is.

The component should become available again once rust-lang/rust#52280 makes its way into a nightly.

japaric, 2018-07-15:
The component is available again.

closed 2018-07-15 by japaric