rust-embedded/cargo-binutils #17: error: unable to get target for 'riscv32imac-unknown-none-elf'

japaric, 2018-08-08:
### STR

``` console
$ cat src/lib.rs
```

``` rust
#![no_std]

#[no_mangle]
pub fn foo(x: u32, y: u32) -> u32 {
    x + y
}
```

``` console
$ cargo build --target riscv32imac-unknown-none-elf --release
$ cargo objdump --target riscv32imac-unknown-none-elf -v -- -d target/riscv32imac-unknown-none-elf/release/libfoo.rlib
"/home/japaric/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump" "-triple" "riscv32imac-unknown-none-elf" "-d" "target/riscv32imac-unknown-none-elf/release/libfoo.rlib"

target/riscv32imac-unknown-none-elf/release/libfoo.rlib(foo-5f01dd28752961b9.foo0-d59093d565f158ef2f92ae2d0a10a47.rs.rcgu.o):     file format ELF32-riscv

/home/japaric/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objdump: 'foo-5f01dd28752961b9.foo0-d59093d565f158ef2f92ae2d0a10a47.rs.rcgu.o': can't find target: : error: unable to get target for 'riscv32imac-unknown-none-elf', see --version and --triple.
.
```

The problem is that the triple passed to llvm-objdump is not a valid LLVM triple; `riscv32imac` is not a valid LLVM architecture name.

A solution to this problem would be to use pass `-arch-name`, instead of `-triple`, to llvm-objdump. The argument to pass would then have to be a valid LLVM architecture (see `cargo objdump -- -version`) so we would need to map the Rust architecture (see `rustc --target $T --print cfg`) into a LLVM architecture. We have to be careful with the `thumbv*` targets which have `target_arch: "arm"` but actually map to "thumb" LLVM architecture.

japaric, 2018-08-13:
cc @rust-embedded/riscv

danc86, 2018-08-13:
Makes me wish we had kept the rust arch name as "riscv" instead of "riscv32" ... but there is no nice solution here, the GNU toolchain insists it is "riscv32" or "riscv64" not "riscv". :-(

japaric, 2018-08-25:
Also working on this while I work on #15

dvc94ch, 2018-08-26:
I noticed readelf is missing, how come?

japaric, 2018-08-26:
@dvc94ch probably because no one asked for it. You can send a PR to add it. You can use rust-lang/rust#51962 as a reference.

dvc94ch, 2018-08-26:
Good to know. Put it on my Todo list, thanks.

japaric, 2018-08-26:
PR #19 fixes this

commit 85bcdc6e (2018-08-28): objdump: use -arch-name instead of -triple

infer the arch-name from the output of `rustc --print cfg` is more robust than
just using the target name for the -triple flag.

fixes #17
commit e1a58213 (2018-08-28): Merge #19

19: objdump: use -arch-name instead of -triple r=therealprof a=japaric

infer the arch-name from the output of `rustc --print cfg` is more robust than
just using the target name for the -triple flag.

fixes #17

r? @rust-embedded/tools (anyone)
cc @danc86
PR #19 (2018-08-28): objdump: use -arch-name instead of -triple
infer the arch-name from the output of `rustc --print cfg` is more robust than
just using the target name for the -triple flag.

fixes #17

r? @rust-embedded/tools (anyone)
cc @danc86
closed 2018-08-28 by bors[bot]