rust-embedded/cargo-binutils #18: size detects .rodata as .data

korken89, 2018-08-12:
When using the `cargo size` command it is misinterpreting the `.rodata` section as `.data`. Or am I using it incorrectly?

`cargo size` output:
```
$ cargo size -- -B target/thumbv7em-none-eabihf/release/trustflight 
   text    data     bss     dec     hex filename
    292     508       0     800     320 target/thumbv7em-none-eabihf/release/trustflight
```

`arm-none-eabi-size` output (correct):
```
$ arm-none-eabi-size -B target/thumbv7em-none-eabihf/release/trustflight
   text	   data	    bss	    dec	    hex	filename
    800	      0	      0	    800	    320	target/thumbv7em-none-eabihf/release/trustflight
```

japaric, 2018-08-17:
> Or am I using it incorrectly?

i don't think so, I see the same. This could be a bug in LLVM, or perhaps we ought to be using some flag to get the output we expect. In the meantime, I'm using `cargo size -- -A -x (..)` which shows the size of each section (.text, .rodata, .bss, .data, .vector_table, etc).

korken89, 2018-08-17:
Thanks, then I know it's not in my side :)

japaric, 2018-08-27:
I reported this as a LLVM bug: https://bugs.llvm.org/show_bug.cgi?id=38723

therealprof, 2020-04-14:
Seems to be fixed. Closing now.

closed 2020-04-14 by therealprof