rust-embedded/cargo-binutils #21: Wrong output for armeb (big endian ARM) binaries

japaric, 2018-08-27:
### STR

``` console
$ cargo new --lib foo && cd $_

$ cat >src/lib.rs <<'EOF'
#![no_std]
#[no_mangle]
pub fn foo(x: u32, y: u32) -> u32 {
    x + y
}
EOF

$ cargo objdump --lib --release --target armebv7r-none-eabi -- -d
/home/japaric/tmp/foo/target/armebv7r-none-eabi/release/libfoo.rlib(foo-5c3d4a3cd43899d6.foo.axz7xdhy-cgu.0.rcgu.o):    file format ELF32-arm-big

Disassembly of section .text.foo:
foo:
       0:       e0 81 00 00     andeq   r8, r0, r0, ror #3
       4:       e1 2f ff 1e     cdpne   p15, #15, c2, c15, c1, #7
```

Compare the output to the output of GNU objdump

``` console
$ arm-none-eabi-objdump -d target/armebv7r-none-eabi/release/libfoo.rlib
foo-5c3d4a3cd43899d6.foo.axz7xdhy-cgu.0.rcgu.o:     file format elf32-bigarm

Disassembly of section .text.foo:

00000000 <foo>:
   0:   e0810000        add     r0, r1, r0
   4:   e12fff1e        bx      lr
```

This is a bug in llvm-objdump. Reported as https://bugs.llvm.org/show_bug.cgi?id=38721

### Meta

I used cargo-binutils v0.1.2 (currently unreleased) and tomorrow's nightly so I can't include sensible version information atm.

cc @paoloteti

paoloteti, 2018-08-28:
Thanks for the feedback. I think `llvm-objdump` is not stable enough and Arm Compiler still use `armdis` (at least in the version I use day-by-day) . BTW is this an `armeb` issue or a generic 'big-endian' issue? `thumbeb` seems to have the same issue on a simple `foo.c`.

japaric, 2018-08-28:
I have only tested armeb. If you can provide me a binary blob that shows the problem on the thumbeb backend then I can include it in the LLVM bug report (or you comment there yourself).

paoloteti, 2018-08-29:
Simple foo.c
```
int foo(int a, int b) { return a + b; }
```
```
arm-none-eabi-gcc -O2 -mthumb -c -mbig-endian foo.c -o foo_thumbeb.o
arm-none-eabi-gcc -O2 -mthumb -c foo.c -o foo_thumb.o

```
_GNU objdump (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 2.30.0.20180329_ report:

```
arm-none-eabi-objdump -d foo_*

foo_thumbeb.o:     file format elf32-bigarm

Disassembly of section .text:

00000000 <foo>:
   0:	1840      	adds	r0, r0, r1
   2:	4770      	bx	lr
```

```
foo_thumb.o:     file format elf32-littlearm

Disassembly of section .text:

00000000 <foo>:
   0:	1840      	adds	r0, r0, r1
   2:	4770      	bx	lr
```
Using `llvm-objdump` from `rustc 1.30.0-nightly (7061b2775 2018-08-28)`

```
./llvm-objdump -triple=thumb -d /tmp/foo_thumb.o 
/tmp/foo_thumb.o:	file format ELF32-arm-little

Disassembly of section .text:
foo:
       0:	40 18 	adds	r0, r0, r1
       2:	70 47 	bx	lr
```

```
./llvm-objdump -triple=thumbeb -d /tmp/foo_thumbeb.o 
/tmp/foo_thumbeb.o:	file format ELF32-arm-big

subsection length greater than section length
subsection length greater than section length
Disassembly of section .text:
foo:
       0:	18 40 	ands	r0, r3
       2:	47 70 	strb	r7, [r0, #1]
```

japaric, 2018-09-01:
Thanks for the repro instructions. I have reported the thumbeb problem in the original armeb bug report.

therealprof, 2020-04-14:
No idea whether this has been fixed upstream but it's not really something we can address in `cargo-binutils`. Closing thus.

closed 2020-04-14 by therealprof