rust-embedded/cargo-binutils #32: "'test_app.exe': no such file or directory" dumping thumbv7em from instructions in the book

thejpster, 2018-10-02:
```
user@host $ cat Cargo.toml
[package]
authors = ["user@host.com"]
edition = "2018"
name = "test_app"
version = "0.1.0"

[dependencies]
cortex-m = "0.5.7"
cortex-m-rt = "0.6.3"
cortex-m-semihosting = "0.3.1"
panic-halt = "0.2.0"

# Uncomment for the panic example.
panic-semihosting = "*"

# Uncomment for the allocator example.
# alloc-cortex-m = "0.3.5"

# Uncomment for the device example.
# [dependencies.stm32f30x]
# features = ["rt"]
# version = "0.7.1"

# this lets you use `cargo fix`!
[[bin]]
name = "test_app"
test = false
bench = false

[profile.release]
codegen-units = 1 # better optimizations
debug = true # symbols are nice and they don't increase the size on Flash
lto = true # better optimizations
user@host ~/Documents/rust_training/blank_app $ cargo +beta objdump --bin test_app -- -disassemble -no-show-raw-insn -print-imm-hex
    Finished dev [unoptimized + debuginfo] target(s) in 0.08s
C:\Users\jgp\.rustup\toolchains\beta-x86_64-pc-windows-msvc\lib\rustlib\x86_64-pc-windows-msvc\bin\llvm-objdump.exe: 'test_app.exe': no such file or directory
```

Is this a bug, or have I missed a step? If so, could we make it more obvious what's gone wrong?

Platform: Windows 10
Rust: rustc 1.30.0-beta.9 (a18eb4852 2018-09-30)

Phaiax, 2018-10-15:
I have the same problem. The problem is that the binutils add the `.exe` suffix to the application binary name, which is not present in msvc builds in windows. As a workaroud you can leave out `--bin test_app` and add `target/your-target-triple/debug/test_app` after the two `--` .

closed 2018-10-30 by bors[bot]

GregWoods, 2019-12-09:
i am having the same issue on v 0.1.7 of cargo-binutils

coredump, 2020-03-19:
This problem is back, apparently. Getting it on v 0.1.7

commit d4cf5785 (2018-10-30): fix path inference on windows hosts

also add support for WASM binaries

closes #32
commit ede21401 (2018-10-30): Merge #35

35: fix path inference on windows hosts r=ryankurte a=japaric

also add support for WASM binaries

closes #32

This also prepares a new release

The Windows CI is a shot in the dark. Let's see if it works if not it can be
postponed.
PR #35 (2018-10-30): fix path inference on windows hosts
also add support for WASM binaries

closes #32

This also prepares a new release

The Windows CI is a shot in the dark. Let's see if it works if not it can be
postponed.