rust-embedded/cargo-binutils #41: Support workspaces

roblabla, 2019-01-06:
Right now, cargo nm for instance fails to work when used in a workspace. It errors out with 

```
error: missing field `package`
```

japaric, 2019-02-08:
Could you give us more details about how to reproduce this? Are you calling the tool from the workspace root or somewhere else? What's in your Cargo.toml?

This probably has to be fixed in [cargo-project](https://github.com/japaric/cargo-project) so I'm going to label this as "upstream".

roblabla, 2019-02-08:
Yes, I'm calling the tool from workspace root. Reproduction steps:

```
$ git clone https://github.com/roblabla42/kfs
$ cd kfs
$ cargo nm
error: missing field `package`
```

I'm not sure what cargo nm should do here, but in particular, `cargo nm -- target/debug/kfs_kernel` fails too. I'd expect this to work correctly, as I'm directly passing the path to the ELF, it shouldn't even need to read the Cargo.toml.

Cargo.toml:

```toml
[workspace]
members = ["kernel", "bootstrap", "shell", "libuser", "clock", "sm", "vi", "ahci", "libutils", "libkern"]

[profile.release]
debug = true

[profile.test]
debug = true
```

berkus, 2019-11-15:
Now that cargo-project is fixed, there needs to be a new release of cargo-binutils, right?

therealprof, 2019-11-15:
Nope, recompiling should be enough.

therealprof, 2019-11-15:
However we could do a new release for good measure so people will notice. If you'd like to a PR to bump the version and add the missing commit to the CHANGELOG I'd be happy to run it for you.

berkus, 2019-11-15:
Oh, easier done than said - see #49

berkus, 2019-11-20:
So, with `cargo-binutils` 0.1.7:
```
➤ cargo objcopy --version
cargo-objcopy 0.1.7
```

```
➤ cat Cargo.toml
[workspace]
members = [
    "vesper",
    "crates/dtb",
    "crates/register"
]
```
Results in:
```
➤ make
cp target/aarch64-vesper-metta/release/vesper ./kernel8
cargo objcopy -- --strip-all -O binary target/aarch64-vesper-metta/release/vesper kernel8.img
error: missing field `package`
```

So, not fixed still.

berkowski, 2020-03-30:
Any way to progress this issue forward?  I'm running into the same problem on an embedded project of my own.

roblabla, 2020-03-31:
@berkowski #50 should fix it, but is a very invasive change. I haven't used it in a while though. Maybe try it out and see if it works for you?

berkowski, 2020-03-31:
Seems to work for me

berkus, 2020-04-04:
I had to run it from within workspace members, then it seems to work, so it's fixed. Just requires a bit more hand twisting for it.

closed 2020-04-15 by bors[bot]

berkus, 2020-04-15:
Hurray!