rust-embedded/cargo-binutils #42: `cargo objcopy -O ihex` outputs elf

nickray, 2019-03-11:
According to https://github.com/rust-embedded/debugonomicon/blob/master/src/overview.md, it should be possibly to generate a HEX from an ELF with `cargo objcopy`, similar to how a BIN can be generated, and in analogy with `arm-none-eabi-objcopy`. It seems that passing `-O ihex` however silently outputs an ELF. 

Did this ever work? Should this work?

ryankurte, 2019-05-09:
hi, sorry for the slow reply. i _think_ the command you're looking for is probably `cargo objcopy -- -O ihex`, note the `--` that splits arguments for the objcopy binary or the underlying llvm call.

there's definitely room for improvement with making binutils easier to use :-/

nickray, 2019-05-09:
It's not a syntax or usability issue.

- when I filed this issue/request, the behaviour of `cargo objcopy` (at least for me) was that `-O ihex` _silently_ produced the same output as `-O binary` (namely, a "binary" output, not an "ihex" output)
- today, asking for `-O ihex` complains: `error: Invalid output format: 'ihex'`

I think probably `llvm-objcopy` itself does not support Intel HEX?

pitaj, 2019-08-21:
According to the documentation, objcopy now supports ihex: https://www.llvm.org/docs/CommandGuide/llvm-objcopy.html#supported-formats

How often is this repository updated from mainline llvm?

therealprof, 2020-04-14:
Works just fine now, no idea for how long it did. Closing this issue thus.

closed 2020-04-14 by therealprof