rust-embedded/cargo-binutils #46: Installing on Windows fails with "failed to run custom build command for `backtrace-sys v0.1.31`"

toothbrush7777777, 2019-09-14:
Some crate includes `backtrace-sys`, which tries to compile `libbacktrace` from C source (https://github.com/rust-lang/backtrace-rs/blob/e745627050fde4161235c48fdc859ba4098a8a79/crates/backtrace-sys/build.rs#L23).

```
    Updating crates.io index
  Installing cargo-binutils v0.1.6
   Compiling proc-macro2 v0.4.30
   Compiling unicode-xid v0.1.0
   Compiling libc v0.2.62
   Compiling cc v1.0.45
   Compiling syn v0.15.44
   Compiling proc-macro2 v1.0.3
   Compiling winapi-x86_64-pc-windows-gnu v0.4.0
   Compiling unicode-xid v0.2.0
   Compiling failure_derive v0.1.5
   Compiling winapi v0.3.8
   Compiling serde v1.0.100
   Compiling memchr v2.2.1
   Compiling syn v1.0.5
   Compiling rustc-demangle v0.1.16
   Compiling cfg-if v0.1.9
   Compiling bitflags v1.1.0
   Compiling unicode-width v0.1.6
   Compiling semver-parser v0.7.0
   Compiling regex v0.2.11
   Compiling lazy_static v1.4.0
   Compiling ucd-util v0.1.5
   Compiling strsim v0.8.0
   Compiling vec_map v0.8.1
   Compiling utf8-ranges v1.0.4
   Compiling textwrap v0.11.0
   Compiling semver v0.9.0
   Compiling thread_local v0.3.6
   Compiling regex-syntax v0.5.6
   Compiling backtrace-sys v0.1.31
   Compiling rustc_version v0.2.3
   Compiling quote v0.6.13
   Compiling quote v1.0.2
   Compiling aho-corasick v0.6.10
error: failed to run custom build command for `backtrace-sys v0.1.31`

Caused by:
  process didn't exit successfully: `C:\Users\USER\AppData\Local\Temp\cargo-installGpuluj\release\build\backtrace-sys-a9436706a9cb41a7\build-script-build` (exit code: 1)
--- stdout
cargo:rustc-cfg=rbt
TARGET = Some("x86_64-pc-windows-gnu")
OPT_LEVEL = Some("3")
HOST = Some("x86_64-pc-windows-gnu")
CC_x86_64-pc-windows-gnu = None
CC_x86_64_pc_windows_gnu = None
HOST_CC = None
CC = None
CFLAGS_x86_64-pc-windows-gnu = None
CFLAGS_x86_64_pc_windows_gnu = None
HOST_CFLAGS = None
CFLAGS = None
CRATE_CC_NO_DEFAULTS = None
CARGO_CFG_TARGET_FEATURE = Some("fxsr,mmx,sse,sse2")
running: "gcc.exe" "-O3" "-ffunction-sections" "-fdata-sections" "-m64" "-I" "src/libbacktrace" "-I" "C:\\Users\\USER\\AppData\\Local\\Temp\\cargo-installGpuluj\\release\\build\\backtrace-sys-809663e299ff9da7\\out" "-fvisibility=hidden" "-DBACKTRACE_SUPPORTED=1" "-DBACKTRACE_USES_MALLOC=1" "-DBACKTRACE_SUPPORTS_THREADS=0" "-DBACKTRACE_SUPPORTS_DATA=0" "-DHAVE_DL_ITERATE_PHDR=1" "-D_GNU_SOURCE=1" "-D_LARGE_FILES=1" "-Dbacktrace_full=__rbt_backtrace_full" "-Dbacktrace_dwarf_add=__rbt_backtrace_dwarf_add" "-Dbacktrace_initialize=__rbt_backtrace_initialize" "-Dbacktrace_pcinfo=__rbt_backtrace_pcinfo" "-Dbacktrace_syminfo=__rbt_backtrace_syminfo" "-Dbacktrace_get_view=__rbt_backtrace_get_view" "-Dbacktrace_release_view=__rbt_backtrace_release_view" "-Dbacktrace_alloc=__rbt_backtrace_alloc" "-Dbacktrace_free=__rbt_backtrace_free" "-Dbacktrace_vector_finish=__rbt_backtrace_vector_finish" "-Dbacktrace_vector_grow=__rbt_backtrace_vector_grow" "-Dbacktrace_vector_release=__rbt_backtrace_vector_release" "-Dbacktrace_close=__rbt_backtrace_close" "-Dbacktrace_open=__rbt_backtrace_open" "-Dbacktrace_print=__rbt_backtrace_print" "-Dbacktrace_simple=__rbt_backtrace_simple" "-Dbacktrace_qsort=__rbt_backtrace_qsort" "-Dbacktrace_create_state=__rbt_backtrace_create_state" "-Dbacktrace_uncompress_zdebug=__rbt_backtrace_uncompress_zdebug" "-Dmacho_get_view=__rbt_macho_get_view" "-Dmacho_symbol_type_relevant=__rbt_macho_symbol_type_relevant" "-Dmacho_get_commands=__rbt_macho_get_commands" "-Dmacho_try_dsym=__rbt_macho_try_dsym" "-Dmacho_try_dwarf=__rbt_macho_try_dwarf" "-Dmacho_get_addr_range=__rbt_macho_get_addr_range" "-Dmacho_get_uuid=__rbt_macho_get_uuid" "-Dmacho_add=__rbt_macho_add" "-Dmacho_add_symtab=__rbt_macho_add_symtab" "-Dmacho_file_to_host_u64=__rbt_macho_file_to_host_u64" "-Dmacho_file_to_host_u32=__rbt_macho_file_to_host_u32" "-Dmacho_file_to_host_u16=__rbt_macho_file_to_host_u16" "-o" "C:\\Users\\USER\\AppData\\Local\\Temp\\cargo-installGpuluj\\release\\build\\backtrace-sys-809663e299ff9da7\\out\\src/libbacktrace\\alloc.o" "-c" "src/libbacktrace/alloc.c"

--- stderr

error occurred: Failed to find tool. Is `gcc.exe` installed? (see https://github.com/alexcrichton/cc-rs#compile-time-requirements for help)

warning: build failed, waiting for other jobs to finish...
error: failed to compile `cargo-binutils v0.1.6`, intermediate artifacts can be found at `C:\Users\USER\AppData\Local\Temp\cargo-installGpuluj`

Caused by:
  build failed
```

therealprof, 2020-04-14:
Should work just fine now, please open another ticket if it still doesn't work for you.

closed 2020-04-14 by therealprof

DJ-Dav, 2023-03-01:
running cargo install -f cargo-binutils causes the following: 
  Updating crates.io index
  Installing cargo-binutils v0.3.6
   Compiling proc-macro2 v1.0.51
   Compiling quote v1.0.23
   Compiling unicode-ident v1.0.6
   Compiling syn v1.0.109
   Compiling serde_derive v1.0.152
error: failed to run custom build command for "quote v1.0.23"

Caused by:
  could not execute process "ME\AppData\Local\Temp\cargo-installE7ipUF\release\build\quote-0cc442527a6b4d49\build-script-build" (never executed)

Caused by:
  The system cannot find the file specified. (os error 2)
warning: build failed, waiting for other jobs to finish...
error: failed to compile "cargo-binutils v0.3.6", intermediate artifacts can be found at "ME\AppData\Local\Temp\cargo-installE7ipUF"

Emilgardis, 2023-03-01:
@DJ-Dav That error suggests something wrong with your toolchain, but it's as you suggest not a problem from `cargo-binutils` but with cargo.

I'd recommend opening an issue on https://github.com/rust-lang/cargo or chiming in on https://github.com/rust-lang/cargo/issues/9902 which seems related (and with possible solutions)