rust-embedded/cargo-binutils #52: `rust-lld` is badly behaved and should possibly be renamed

jonas-schievink, 2020-02-18:
`rust-lld` can shadow the actual linker invoked by rustc, which is usually not the desired behavior, so it should be renamed.

In addition to that:

* `rust-lld` only looks in the sysroot of the default toolchain, which might not be the toolchain invoking it. This can result in nightly rustc using stable `rust-lld`, which is not a supported configuration.
* `rust-lld` also exits with a successful status code when it cannot find the actual `rust-lld` binary. This results in rustc thinking the linker succeeded, without actually producing any artifact. Cargo doesn't handle this case properly, and will try to run something that doesn't exist.

therealprof, 2020-02-18:
Funny, I just noticed the same some hours ago.

therealprof, 2020-04-15:
Caused by #48 .

@jonas-schievink 

> rust-lld only looks in the sysroot of the default toolchain, which might not be the toolchain invoking it. This can result in nightly rustc using stable rust-lld, which is not a supported configuration.

How does this normally work? When invoked from `cargo` this works but where does a binary like `rust-lld` get the correct sysroot from when invoked without `cargo`?

```
cargo +nightly run --bin rust-lld
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s
     Running `target/debug/rust-lld`
[src/lib.rs:539] search_tool(tool) = Ok(
    "/Users/egger/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/bin/rust-lld",
)
lld is a generic driver.
Invoke ld.lld (Unix), ld64.lld (macOS), lld-link (Windows), wasm-ld (WebAssembly) instead
```

```
# cargo run --bin rust-lld
    Finished dev [unoptimized + debuginfo] target(s) in 0.15s
     Running `target/debug/rust-lld`
[src/lib.rs:539] search_tool(tool) = Ok(
    "/Users/egger/.rustup/toolchains/stable-x86_64-apple-darwin/lib/rustlib/x86_64-apple-darwin/bin/rust-lld",
)
lld is a generic driver.
Invoke ld.lld (Unix), ld64.lld (macOS), lld-link (Windows), wasm-ld (WebAssembly) instead
```

> rust-lld also exits with a successful status code when it cannot find the actual rust-lld binary. This results in rustc thinking the linker succeeded, without actually producing any artifact. Cargo doesn't handle this case properly, and will try to run something that doesn't exist.

Can't reproduce this. Have an example?

jonas-schievink, 2020-04-15:
I suppose you could try `rustc --print sysroot`... hmm, but that's also the default rustc, not the one invoking rust-lld. Maybe try to check if there's something in the environment that tells you? Or perhaps the solution is just to rename the rust-lld wrapper so the compiler doesn't invoke it by default?

> Can't reproduce this. Have an example?

Its implementation is just

https://github.com/rust-embedded/cargo-binutils/blob/227f7138222649591616b0d82f6491e5a93eb96d/src/bin/rust-lld.rs#L6-L9

The `Err` case is what gets executed when the rust-lld executable isn't found (which means that `Command::status()` returns `Err`).

therealprof, 2020-04-15:
> The Err case is what gets executed when the rust-lld executable isn't found (which means that Command::status() returns Err).

🤦‍♂️ We can fix that. Thanks for pointing it out.

ZeroErrors, 2020-04-23:
@therealprof I just pushed some local changes I've been experimenting with that includes a fix for this issue, however problem is it also has quite a lot of other changes.
https://github.com/ZeroErrors/cargo-binutils/tree/experimental

I can probably extract just the changes I made to `forward` and PR that.
https://github.com/ZeroErrors/cargo-binutils/blob/7359932b6ee56a8e05304aa13068008cbe8f6369/src/tool.rs#L67-L93
https://github.com/ZeroErrors/cargo-binutils/blob/7359932b6ee56a8e05304aa13068008cbe8f6369/src/bin/rust-lld.rs

therealprof, 2020-04-23:
@ZeroErrors Sounds excellent, I'd be happy to take a look at a PR.

closed 2020-05-18 by bors[bot]