rust-embedded/cargo-binutils #63: cargo-metadata pipe use creates aweful error messages

therealprof, 2020-04-14:
When an error arises during `cargo-metadata` parsing, broken pipes and other nasties may appear, e.g.:
```
# cargo size
 Compiling cargo-binutils v0.2.0 (/Users/egger/OSS/cargo-binutils)
error: Can only have one matching artifact but found several=====>     ] 70/76: cargo-readobj(bin), rust-objcopy(bin), rust-ar(bin), rust-ld(bin), cargo-size(bin), rust-lld(bin)
Building [===================================================>     ] 70/76: cargo-readobj(bin), rust-objcopy(bin), rust-ar(bin), rust-ld(bin), cargothread 'main' panicked at 'failed printing to stdout: Broken pipe (os error 32)', src/libstd/io/stdio.rs:805:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

ZeroErrors, 2020-04-15:
The mixed lines are caused by the cargo command outputting to stderr which by default is using `Stdio::inherit()` (the following line for example: ```  Compiling cargo-binutils v0.2.0 (/Users/egger/OSS/cargo-binutils)```)

The broken pipe can be fixed by killing the process [[1]](#patch), however this can still cause issues because of the previously printed lines (or partial lines) as well as output on stderr can cause the formatting to break.
So a better option could be to just let the `cargo build` finish and then error unless there is a safe way to communicate to cargo that we want to stop the build early.

<a id="patch" />[1]
```patch
diff --git a/src/lib.rs b/src/lib.rs
index 87e4e80..cc65ed8 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -278,6 +278,7 @@ fn determine_artifact(
             Message::CompilerArtifact(artifact) => {
                 if artifact.package_id == package_id && build_type.matches(&artifact) {
                     if wanted_artifact.is_some() {
+                        child.kill()?;
                         bail!("Can only have one matching artifact but found several");
                     }

```

therealprof, 2020-04-15:
I agree in principle however when I tried parsing the metadata at once, the Windows CI wasn't happy and timed out with pipe problems, cf. https://travis-ci.org/github/rust-embedded/cargo-binutils/builds/673650482

therealprof, 2020-04-15:
At some point we should also properly catch the error and turn it into a return code rather than panicing. Will open an issue for that. ;)

ZeroErrors, 2020-04-15:
From a quick glance at the changes I think the reason it failed is because of the order of operation.
```
let messages = parse_messages(stdout);
let status = child.wait()?;
..
for message in messages {
```

I assume the `MessageIterator` returned from `parse_messages` only parses data as the iterator is iterated so it blocks stdout until we get to the `for`, however since the `.wait()` happens before that we also block on the process existing. I assume this is an issue only on windows due to how buffering stdout works.

I think changing it to something like this should solve that problem.
```rust
let messages = parse_messages(stdout).collect::<Vec<_>>()
```

ZeroErrors, 2020-04-15:
Made a quick change to test it out and it seems to be working for me on windows. Branch here: [ZeroErrors:collect-messages ](https://github.com/ZeroErrors/cargo-binutils/tree/collect-messages)

therealprof, 2020-04-15:
I think you're right. Let's do this then.

closed 2020-04-15 by bors[bot]