rust-embedded/cargo-binutils #66: Don't panic!

therealprof, 2020-04-15:
As seen in #63 there're code paths which can lead to a `panic!` rather than an orderly error handling and reflection in an appropriate error code. Let's fix those!

ZeroErrors, 2020-05-23:
I looked over the code (as of the changes in the PR https://github.com/rust-embedded/cargo-binutils/pull/74) to see what code can panic and as far as I can see the only places that can panic would be due to a change in the build-cache structure or other abnormal issues caused by cargo or the OS. So I think its safe to close this issue.

---

https://github.com/rust-embedded/cargo-binutils/blob/8a6da67d395474832fea3b6cc620c6984529206c/src/lib.rs#L69-L71
Should only be possible if the [build-cache](https://doc.rust-lang.org/cargo/guide/build-cache.html) structure changed.

---

https://github.com/rust-embedded/cargo-binutils/blob/8a6da67d395474832fea3b6cc620c6984529206c/src/lib.rs#L355-L357
Could only be possible if `artifact.executable` somehow pointed to the root `/` which would mean cargo gave us invalid data.

therealprof, 2020-05-23:
Indeed this looks **a lot better** now than back in April. Happy to close this off now.

closed 2020-05-23 by therealprof