rust-embedded/cargo-binutils #69: cargo <binutil> with --lib panics

idoasher, 2020-04-23:
here for example the output of `cargo strip --lib minigrep --release`:
```
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', /home/ido/.cargo/registry/src/github.com-1ecc6299db9ec823/car
go-binutils-0.2.0/src/lib.rs:232:24
stack backtrace:
   0:     0x7fe44b5bda44 - backtrace::backtrace::libunwind::trace::h90669f559fb267f0
                               at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/libunwind.rs:88
   1:     0x7fe44b5bda44 - backtrace::backtrace::trace_unsynchronized::hffde4e353d8f2f9a
                               at /cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.40/src/backtrace/mod.rs:66
   2:     0x7fe44b5bda44 - std::sys_common::backtrace::_print_fmt::heaf44068b7eaaa6a
                               at src/libstd/sys_common/backtrace.rs:77
   3:     0x7fe44b5bda44 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h88671019cf081de2
                               at src/libstd/sys_common/backtrace.rs:59
   4:     0x7fe44b5df99c - core::fmt::write::h4e6a29ee6319c9fd
                               at src/libcore/fmt/mod.rs:1052
   5:     0x7fe44b5ba857 - std::io::Write::write_fmt::hf06b1c86d898d7d6
                               at src/libstd/io/mod.rs:1426
   6:     0x7fe44b5bfb25 - std::sys_common::backtrace::_print::h404ff5f2b50cae09
                               at src/libstd/sys_common/backtrace.rs:62
   7:     0x7fe44b5bfb25 - std::sys_common::backtrace::print::hcc4377f1f882322e
                               at src/libstd/sys_common/backtrace.rs:49
   8:     0x7fe44b5bfb25 - std::panicking::default_hook::{{closure}}::hc172eff6f35b7f39
                               at src/libstd/panicking.rs:204
   9:     0x7fe44b5bf811 - std::panicking::default_hook::h7a68887d113f8029
                               at src/libstd/panicking.rs:224
  10:     0x7fe44b5c012a - std::panicking::rust_panic_with_hook::hb7ad5693188bdb00
                               at src/libstd/panicking.rs:472
  11:     0x7fe44b5bfd10 - rust_begin_unwind
                               at src/libstd/panicking.rs:380
  12:     0x7fe44b5de4b1 - core::panicking::panic_fmt::hb1f3e14b86a3520c
                               at src/libcore/panicking.rs:85
  13:     0x7fe44b5de3fd - core::panicking::panic::hcdc9f0ba8d71d265
                               at src/libcore/panicking.rs:52
  14:     0x7fe44b43d04e - cargo_binutils::run::h139278197bcb7147
  15:     0x7fe44b434cb1 - cargo_strip::main::h70ef9e58a732eee1
  16:     0x7fe44b434e63 - std::rt::lang_start::{{closure}}::hea9b0dbe894299c8
  17:     0x7fe44b5bfbf3 - std::rt::lang_start_internal::{{closure}}::hb26e39676675046f
                               at src/libstd/rt.rs:52
  18:     0x7fe44b5bfbf3 - std::panicking::try::do_call::he4701ab6e48d80c0
                               at src/libstd/panicking.rs:305
  19:     0x7fe44b5c6a57 - __rust_maybe_catch_panic
                               at src/libpanic_unwind/lib.rs:86
  20:     0x7fe44b5c05d0 - std::panicking::try::hd3de25f3cb7024b8
                               at src/libstd/panicking.rs:281
  21:     0x7fe44b5c05d0 - std::panic::catch_unwind::h86c02743a24e3d92
                               at src/libstd/panic.rs:394
  22:     0x7fe44b5c05d0 - std::rt::lang_start_internal::h9cf8802361ad86c2
                               at src/libstd/rt.rs:51
  23:     0x7fe44b434d92 - main
  24:     0x7fe44ac21b97 - __libc_start_main
  25:     0x7fe44b4349ea - _start
  26:                0x0 - <unknown>
```

it panics for each binutil command I tried (strip, objdump, objcopy, and readobj).
occured on multiple projects.

After a while I figured out that running  the commands without --lib (aka  `cargo strip minigrep --release`) works as expected.

ZeroErrors, 2020-04-23:
This was fixed by https://github.com/rust-embedded/cargo-binutils/pull/59 just waiting on a new release.
If you want to use the tools on libs before then you will need to install from source, (clone the repo and run `cargo install --path .`)

therealprof, 2020-04-23:
@ZeroErrors I'd like to fix the reported error codes as mentioned in #52 before doing the release.

therealprof, 2020-05-23:
#52 was addressed so let's close this now.

closed 2020-05-23 by therealprof