rust-embedded/cargo-binutils #71: v0.2.0: How to use xbuild with objdump?

rjoleary, 2020-04-27:
We used to xbuild our binary for an embedded system like:

```
cargo xbuild
cargo objdump -- -O binary inputfile outputfile
```

This now gives the error: `llvm-objcopy: error: too many positional arguments`
This started happening with v0.2.0. I suspect it is the switch to "cargo-metadata".

One fix we tried is to use a single "cargo objdump" command:

```
cargo objdump --bin inputbinary -- -O binary outputfile
```

However this way of invoking objdump builds the binary and without xbuild.

Is still possible to get the old behavior where there is no default `--bin`?

therealprof, 2020-04-27:
Why not use llvm-objdump directly if you don't want cargo to be involved?

rjoleary, 2020-04-27:
Using rust-objdump works beautifully!
Thank you!

closed 2020-04-27 by rjoleary