rust-embedded/cargo-binutils #81: cargo size error: Can only have one matching artifact but found several

korken89, 2020-07-09:
Hi,

When running `cargo size --release` (cargo-binutils v0.3.0) on a fresh repository I get the following error:

```
error: Can only have one matching artifact but found several
```

However checking the `target` folders there is only one artifact (`app` in my case):

```
thumbv7em-none-eabihf/debug/deps/app-a3b7f7101f0da200.d
thumbv7em-none-eabihf/debug/deps/app-f6b81f508609ad5e.d
thumbv7em-none-eabihf/debug/deps/libapp-a3b7f7101f0da200.rmeta
thumbv7em-none-eabihf/debug/deps/libapp-f6b81f508609ad5e.rmeta
thumbv7em-none-eabihf/release/app
thumbv7em-none-eabihf/release/app.d
thumbv7em-none-eabihf/release/deps/app-1f4505e64d40502e.d
thumbv7em-none-eabihf/release/deps/app-8b955e72da6ee189
thumbv7em-none-eabihf/release/deps/app-8b955e72da6ee189.d
thumbv7em-none-eabihf/release/deps/libapp-1f4505e64d40502e.rlib
thumbv7em-none-eabihf/release/deps/libapp-1f4505e64d40502e.rmeta
thumbv7em-none-eabihf/release/libapp.d
thumbv7em-none-eabihf/release/libapp.rlib
```

What seems to be happening is that it does not realize it should look for the binary.
Running the following gives the expected result: `cargo size --release --bin app`
And I have the `[[bin]]` section in the Cargo.toml, which was working fine before (not sure which version of `cargo-binutils`).

therealprof, 2020-07-09:
Hm... can't reproduce. Neither with nor without `[[bin]]`.

```
# cargo size --release
    Finished release [optimized + debuginfo] target(s) in 0.03s
   text	   data	    bss	    dec	    hex	filename
  15308	      0	      4	  15312	   3bd0	app
```

korken89, 2020-07-10:
Weird it seems @thalesfragoso also had the same issue as me.
Not sure how to not make it happen, because for me it happens on all repositories now.

therealprof, 2020-07-10:
Can you point to a public repo where this happens? Maybe there's a subtle difference in setup.

DerFetzer, 2020-07-24:
I am having the same problem on one of my [repos](https://github.com/DerFetzer/open-cleverpad).
[Another one](https://github.com/DerFetzer/stm32-ccs811) is working fine.
I am using cargo-binutils v0.2.0.

therealprof, 2020-07-25:
> I am using cargo-binutils v0.2.0.

And with 0.3.0?

therealprof, 2020-07-25:
Hah! Found the problem. That darn `build.rs` causes an additional artifact of type "custom-build", that's why cargo complains about multiple artifacts.

korken89, 2020-07-25:
Awesome job!

therealprof, 2020-07-25:
That doesn't mean I have any idea how to properly fix it, yet. ;) Should we just ignore custom build artifacts? Not sure if people are not using that for real...

commit 7d137e4c (2020-07-29): Filter "custom-build" artifact kinds from automatching

If a project uses build.rs, cargo will add a "custom-build" artifact
kind to the build which is confusing information because we can only use
concrete compiled "bin" or "example" binaries for binutils, so let's
only admit real "bin" or "example" kinds into the auto matcher.

Fixes #81
commit 2c39f5c9 (2020-07-29): Merge #82

82: Filter "custom-build" artifact kinds from automatching r=adamgreig a=therealprof

If a project uses build.rs, cargo will add a "custom-build" artifact
kind to the build which is confusing information because we can only use
concrete compiled "bin" or "example" binaries for binutils, so let's
only admit real "bin" or "example" kinds into the auto matcher.

Fixes #81
PR #82 (2020-07-29): Filter "custom-build" artifact kinds from automatching
If a project uses build.rs, cargo will add a "custom-build" artifact
kind to the build which is confusing information because we can only use
concrete compiled "bin" or "example" binaries for binutils, so let's
only admit real "bin" or "example" kinds into the auto matcher.

Fixes #81
closed 2020-07-29 by bors[bot]