rust-embedded/cargo-binutils #96: no memory region specified for section '.eh_frame' (Ubuntu 16 i686)

advancedwebdeveloper, 2020-11-10:
Hi.
I believe you know the cause of [this](https://github.com/riscv-rust/longan-nano/issues/15) bug.

ryankurte, 2020-11-10:
i am _not at all_ sure this is it but, usually with ARM we specify the GNU/gcc linker rather than LLD with `.cargo/config`:

```
[target.armv7-unknown-linux-gnueabihf]
linker = "arm-linux-gnueabihf-gcc"
```

have you tried building with the appropriate GNU linker for riskv?

also may be related to: https://github.com/rust-embedded/wg/issues/36#issuecomment-325167189

advancedwebdeveloper, 2020-11-12:
> ~/Desktop/rust-sipeed-longan-nano-master$ cargo build
>    Compiling r0 v0.2.2
>    Compiling panic-abort v0.3.1
>    Compiling rust-sipeed-longan-nano v0.1.0 (/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master)
> warning: unnecessary parentheses around assigned value
>  --> src/main.rs:8:25
>   |
> 8 | const rcu_apb2en: u32 = (0x4002_1000 + 0x18);
>   |                         ^^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
>   |
>   = note: `#[warn(unused_parens)]` on by default
> 
> warning: unnecessary parentheses around assigned value
>   --> src/main.rs:10:25
>    |
> 10 | const gpioa_ctl0: u32 = (0x4001_0800 + 0x0);
>    |                         ^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
> 
> warning: unnecessary parentheses around assigned value
>   --> src/main.rs:11:25
>    |
> 11 | const gpioa_data: u32 = (0x4001_0800 + 0xc);
>    |                         ^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
> 
> warning: unused import: `panic_abort`
>   --> src/main.rs:13:5
>    |
> 13 | use panic_abort;
>    |     ^^^^^^^^^^^
>    |
>    = note: `#[warn(unused_imports)]` on by default
> 
> warning: constant `rcu_apb2en` should have an upper case name
>  --> src/main.rs:8:7
>   |
> 8 | const rcu_apb2en: u32 = (0x4002_1000 + 0x18);
>   |       ^^^^^^^^^^ help: convert the identifier to upper case: `RCU_APB2EN`
>   |
>   = note: `#[warn(non_upper_case_globals)]` on by default
> 
> warning: constant `gpioa_ctl0` should have an upper case name
>   --> src/main.rs:10:7
>    |
> 10 | const gpioa_ctl0: u32 = (0x4001_0800 + 0x0);
>    |       ^^^^^^^^^^ help: convert the identifier to upper case: `GPIOA_CTL0`
> 
> warning: constant `gpioa_data` should have an upper case name
>   --> src/main.rs:11:7
>    |
> 11 | const gpioa_data: u32 = (0x4001_0800 + 0xc);
>    |       ^^^^^^^^^^ help: convert the identifier to upper case: `GPIOA_DATA`
> 
> error: linking with `riscv32-unknown-elf-gcc` failed: exit code: 1
>   |
>   = note: "riscv32-unknown-elf-gcc" "-L" "/home/oceanfish81/.rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.19pbfx1e7ppiz22i.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.1gxn96a4ynn6krtc.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.1rrq47tcmo7r1d7q.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.2oywxm1qwyrau2j5.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.36wqpyrxaht0bqyw.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.4g879aqcx3v63xcn.rcgu.o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.hpztk5fc5xtxkyj.rcgu.o" "-o" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5" "-Wl,--gc-sections" "-nodefaultlibs" "-L" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps" "-L" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/debug/deps" "-L" "/home/oceanfish81/.rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib" "-Wl,-Bstatic" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libr0-c85b5ab54aceb5f9.rlib" "-Wl,--start-group" "/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/libpanic_abort-37322ac1e75fee7c.rlib" "/home/oceanfish81/.rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib/librustc_std_workspace_core-1bd0d3673780d693.rlib" "/home/oceanfish81/.rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib/libcore-5aec724f1f867fb1.rlib" "-Wl,--end-group" "/home/oceanfish81/.rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib/libcompiler_builtins-deaf2d068ddf18c0.rlib" "-Tlink.x" "-Wl,-Bdynamic"
>   = note: /usr/local/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.19pbfx1e7ppiz22i.rcgu.o: can't link soft-float modules with double-float modules
>           /usr/local/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed_longan_nano-2df4b3c694b8dfb5.19pbfx1e7ppiz22i.rcgu.o
>           /usr/local/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: /home/oceanfish81/Desktop/rust-sipeed-longan-nano-master/target/riscv32imac-unknown-none-elf/debug/deps/rust_sipeed [... cut ...]
>           /usr/local/lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /home/oceanfish81/.rustup/toolchains/nightly-i686-unknown-linux-gnu/lib/rustlib/riscv32imac-unknown-none-elf/lib/libcompiler_builtins-deaf2d068ddf18c0.rlib(compiler_builtins-deaf2d068ddf18c0.compiler_builtins.7egkskwx-cgu.17.rcgu.o)
>           collect2: error: ld returned 1 exit status
>           
> 
> error: aborting due to previous error; 7 warnings emitted
> 
> error: could not compile `rust-sipeed-longan-nano`
> 
> To learn more, run the command again with --verbose.
>

advancedwebdeveloper, 2020-11-12:
That's with

> [target.riscv32imac-unknown-none-elf]
> linker = "riscv32-unknown-elf-ld"

advancedwebdeveloper, 2020-11-12:
However with 

> [target.riscv32imac-unknown-none-elf]
> linker = "riscv32-unknown-elf-ld"

I didn't get any error:

> $ cargo build
>    Compiling panic-abort v0.3.1
>    Compiling r0 v0.2.2
>    Compiling rust-sipeed-longan-nano v0.1.0 (/home/oceanfish81/Desktop/rust-sipeed-longan-nano-master)
> warning: unnecessary parentheses around assigned value
>  --> src/main.rs:8:25
>   |
> 8 | const rcu_apb2en: u32 = (0x4002_1000 + 0x18);
>   |                         ^^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
>   |
>   = note: `#[warn(unused_parens)]` on by default
> 
> warning: unnecessary parentheses around assigned value
>   --> src/main.rs:10:25
>    |
> 10 | const gpioa_ctl0: u32 = (0x4001_0800 + 0x0);
>    |                         ^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
> 
> warning: unnecessary parentheses around assigned value
>   --> src/main.rs:11:25
>    |
> 11 | const gpioa_data: u32 = (0x4001_0800 + 0xc);
>    |                         ^^^^^^^^^^^^^^^^^^^ help: remove these parentheses
> 
> warning: unused import: `panic_abort`
>   --> src/main.rs:13:5
>    |
> 13 | use panic_abort;
>    |     ^^^^^^^^^^^
>    |
>    = note: `#[warn(unused_imports)]` on by default
> 
> warning: constant `rcu_apb2en` should have an upper case name
>  --> src/main.rs:8:7
>   |
> 8 | const rcu_apb2en: u32 = (0x4002_1000 + 0x18);
>   |       ^^^^^^^^^^ help: convert the identifier to upper case: `RCU_APB2EN`
>   |
>   = note: `#[warn(non_upper_case_globals)]` on by default
> 
> warning: constant `gpioa_ctl0` should have an upper case name
>   --> src/main.rs:10:7
>    |
> 10 | const gpioa_ctl0: u32 = (0x4001_0800 + 0x0);
>    |       ^^^^^^^^^^ help: convert the identifier to upper case: `GPIOA_CTL0`
> 
> warning: constant `gpioa_data` should have an upper case name
>   --> src/main.rs:11:7
>    |
> 11 | const gpioa_data: u32 = (0x4001_0800 + 0xc);
>    |       ^^^^^^^^^^ help: convert the identifier to upper case: `GPIOA_DATA`
> 
> warning: 7 warnings emitted
> 
>     Finished dev [unoptimized + debuginfo] target(s) in 0.95s
>

advancedwebdeveloper, 2020-11-12:
I am able to see the binary file, as described [here](https://pramode.in/2019/10/07/rust-on-riscv-board-sipeed-longan-nano/) - it is just I have my toolchain offers "riscv32-unknown-elf" prefix, for the executibles, instead of "riscv-nuclei-elf-"

ryankurte, 2020-11-12:
ahh `can't link soft-float modules with double-float modules` is familiar, though i can't find the issues i tracked down to solve this. on armhf this is sometimes resolved by adding a linker path to the correct libraries for the ABI, something like this in `.cargo/config` from an armhf project of mine:

``` toml
[target.thumbv7em-none-eabihf]

rustflags = [
   "-C", "linker=arm-none-eabi-gcc",
   "-C", "link-arg=-L/usr/lib/arm-none-eabi/lib/thumb/v7e-m/fpv4-sp/hard/",
]
```

> I am able to see the binary file, as described here - it is just I have my toolchain offers "riscv32-unknown-elf" prefix, for the executibles, instead of "riscv-nuclei-elf-"

great! you should just be able to change file paths then?

advancedwebdeveloper, 2020-11-13:
> ahh `can't link soft-float modules with double-float modules` is familiar, though i can't find the issues i tracked down to solve this. on armhf this is sometimes resolved by adding a linker path to [...]

@lupyuen , could you help here?

Let's ask those who know how to do that properly.

> 
> > I am able to see the binary file, as described here - it is just I have my toolchain offers "riscv32-unknown-elf" prefix, for the executibles, instead of "riscv-nuclei-elf-"
> 
> great! you should just be able to change file paths then?
Not really.
After replacing "-gcc" with "-ld" I got a binary, on the output - but has an oddly small size (a bit larger then 1 KB, compared to 20+ KB for known firmware binaries.)
After downloading via DFU my Longan Nano does not boot even a LED related demo project.
Something is wrong.

CC @zsdonghao @junhuanchen @myuanz @QinYUN575 @btx000 @pcotret @Zepan @MindShow

luojia65, 2020-11-13:
It says `can't link soft-float modules with double-float modules`, what could this 'double-float module' be? There's no float and double extension in rust's `riscv32imac` target.

advancedwebdeveloper, 2020-11-13:
I could try to reproduce again an share all generated artifacts

pcotret, 2020-11-13:
> > After downloading via DFU my Longan Nano does not boot even a LED related demo project.
> > Something is wrong.
> CC @zsdonghao @junhuanchen @myuanz @QinYUN575 @btx000 @pcotret @Zepan @MindShow
The only problem I had related to this was that my Longan Nano was kinda stuck.
https://github.com/sipeed/Longan-DOC/blame/master/en/get_started/blink.md#L104
> Development board hold down the BOOT key, then press the RESET button to restart development board and then release the BOOT button to enter DFU mode.

therealprof, 2020-11-13:
> i am _not at all_ sure this is it but, usually with ARM we specify the GNU/gcc linker rather than LLD with `.cargo/config`:

Who is "we" and why are you doing that?

advancedwebdeveloper, 2020-11-14:
I see that something is rather wrong with the config.

For instance I tried to build https://github.com/pdx-cs-rust/nanoball - but it failed, with my recent ~/.cargo/config file.
Once I removed it (but didn't touch anything inside ./.cargo/config file ) - I was able to build the project.
I see the animation on Longan Nano's LCD screen, right now.

ryankurte, 2020-11-14:
> Who is "we" and why are you doing that?

org project, because inexplicably when you start linking external c libraries the linker seems to break in a variety of horrifying ways / stop being able to infer a bunch of things like this correctly.

> what could this 'double-float module' be

oh yeah this is, v different from what i was thinking.

advancedwebdeveloper, 2020-11-14:
@barafael , experiencing an uncertain situation with your demos:
https://github.com/barafael/rust-sipeed-longan-DAC/issues/1