rust-embedded/cargo-binutils #100: Consider moving away from rustc-cfg?

brightly-salty, 2021-01-17:
It looks like rustc-cfg is unmaintained or at least dormant. There are no commits since October 2018, and it's dependence on failure (despite a PR to fix it) makes `cargo-audit` fail for cargo-binutils and anything that depends on it. 

It might be better to move this functionality into cargo-binutils itself. rustc-cfg/src/lib.rs contains less than 200 lines of code, many of which are comments. If the licenses are compatible, I think it would be beneficial to move this file/code into cargo-binutils so it can continue to receive maintenance and updates.

therealprof, 2021-01-17:
Fair enough. @japaric who's both the original author of the dependency in question and `cargo-binutils` might have an opinion on how we could proceed here.

japaric, 2021-02-15:
Importing rustc-cfg code into this repo seems fine to me. I think there's another crate that does about the same as rustc-cfg but I can't remember the name right now.

Emilgardis, 2023-02-03:
rustc-cfg was updated in https://github.com/rust-embedded/cargo-binutils/pull/128

closed 2023-02-03 by Emilgardis