rust-embedded/cargo-binutils #104: `cargo objcopy` looks like passing strange argument to the objcopy command

fukai-t, 2021-04-18:
I am a beginner of rust. When I run `cargo objcopy --verbose --release --target=... -- ...`, it looks like passing a strange argument to the objcopy command.

The execution log is the following.

```
# cargo objcopy --verbose --release --target=aarch64-unknown-none  -- --strip-all -O binary target/aarch64-unknown-none/release/kernel8 kernel8.img
"/usr/local/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "build" "--release" "--target" "aarch64-unknown-none" "--message-format=json"
    Finished release [optimized] target(s) in 0.01s
"rust-objcopy" "/workspace/rust-on-bare-metal-raspi3-samples/target/aarch64-unknown-none/release/kernel8" "--strip-all" "-O" "binary" "target/aarch64-unknown-none/release/kernel8" "kernel8.img"
/usr/local/rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-objcopy: error: too many positional arguments
#
```

I think `"/workspace/rust-on-bare-metal-raspi3-samples/target/aarch64-unknown-none/release/kernel8"` is a strange argument.

If more information is required to solve this problem, please tell me because I am not sure what information is important for debugging.

therealprof, 2021-04-18:
Hi there.

I think your line:
`cargo objcopy --verbose --release --target=aarch64-unknown-none  -- --strip-all -O binary target/aarch64-unknown-none/release/kernel8 kernel8.img`
should read
`cargo objcopy --verbose --release --target=aarch64-unknown-none  -- --strip-all -O binary kernel8.img`

`cargo-objcopy` will automatically pass in the corrrect binary according to your cargo selection, in your case the default binary.

fukai-t, 2021-04-18:
Thank you for your helpful answer.
I have resolved my problem thank to your answer! (I have fixed a given Makefile).
So I close this issue.

closed 2021-04-18 by fukai-t