rust-embedded/cargo-binutils #112: objcopy produces an empty file

TheButlah, 2021-12-18:
I have executed the following commands, both on Windows 10 and on Rasbian 10. All of them produce an empty file. Is this a bug or am I doing something wrong?
````
cargo objcopy --release --target thumbv7em-none-eabihf --bin main -- -O binary asdf.bin
cargo objcopy --target thumbv7em-none-eabihf --bin main -- -O binary asdf.bin
cargo objcopy --release --target thumbv7em-none-eabihf -- -O binary asdf.bin
cargo objcopy --release --target thumbv7em-none-eabihf -- -O binary asdf.bin
```

My main.rs looks like this:
```rust
#![no_std]
#![no_main]

use panic_semihosting as _;

use cortex_m_rt::entry;
use nrf52840_hal::clocks::Clocks;
use nrf52840_hal::usbd::{UsbPeripheral, Usbd};
use nrf52840_pac::Peripherals;
use usb_device::device::{UsbDeviceBuilder, UsbVidPid};
use usbd_serial::{SerialPort, USB_CLASS_CDC};

#[entry]
fn main() -> ! {
    let periph = Peripherals::take().unwrap();
    let clocks = Clocks::new(periph.CLOCK);
    let clocks = clocks.enable_ext_hfosc();

    let usb_bus = Usbd::new(UsbPeripheral::new(periph.USBD, &clocks));
    let mut serial = SerialPort::new(&usb_bus);

    let mut usb_dev = UsbDeviceBuilder::new(&usb_bus, UsbVidPid(0x16c0, 0x27dd))
        .manufacturer("Fake company")
        .product("Serial port")
        .serial_number("TEST")
        .device_class(USB_CLASS_CDC)
        .max_packet_size_0(64) // (makes control transfers 8x faster)
        .build();

    loop {
        if !usb_dev.poll(&mut [&mut serial]) {
            continue;
        }

        let mut buf = [0u8; 64];

        match serial.read(&mut buf) {
            Ok(count) if count > 0 => {
                // Echo back in upper case
                for c in buf[0..count].iter_mut() {
                    if 0x61 <= *c && *c <= 0x7a {
                        *c &= !0x20;
                    }
                }

                let mut write_offset = 0;
                while write_offset < count {
                    match serial.write(&buf[write_offset..count]) {
                        Ok(len) if len > 0 => {
                            write_offset += len;
                        }
                        _ => {}
                    }
                }
            }
            _ => {}
        }
    }
}

```

adamgreig, 2021-12-18:
Have you set a linker script? Something like this line in a .cargo/config.toml file: https://github.com/rust-embedded/cortex-m-quickstart/blob/master/.cargo/config.toml#L18

Without telling cargo to tell Rust to pass -Tlink.x to the linker, your output elf is generated but doesn't have any sections and so the raw binary is empty.

TheButlah, 2021-12-18:
I didn't do that! Is there documentation on this mysterious linker argument so that I can learn more? In the embedded book perhaps?
Does this have anything to do with memory.x? I saw something about that too but don't know what it is or how to configure it

jacobrosenthal, 2021-12-18:
Theres a very small section on it in the embedded book
https://docs.rust-embedded.org/book/start/hardware.html#configuring

And that cortex-m-quickstart is kind of a canonical form of an embedded project and always a good place to check if you're missing something

TheButlah, 2021-12-18:
The memory.x and -Tlink.x stuff fixed it, thank you!

TheButlah, 2021-12-18:
Any chance this could have some diagnostics for it to make this error more debuggable for people like me?

adamgreig, 2021-12-19:
Hm, it's a good idea but it might be tricky to implement here, since this crate is really just providing a convenient wrapper around the Rust distribution `objcopy` and friends. It might be possible to check for an empty output and print a warning but doing so might mess up workflows that expect the behaviour to match the underlying `objcopy`. Still maybe that's worth it...

I think that ideally, the error would instead be caught earlier, since the real problem is that your elf is bad before objcopy even sees it and for people using the elf directly (with e.g. cargo-flash/cargo-embed/probe-run) a change here won't help them. If `cortex-m-rt` could detect the missing linker script somehow instead that would be ideal.

TheButlah, 2021-12-20:
ok, I guess I should close this then

closed 2021-12-20 by TheButlah