rust-embedded/cargo-binutils #113: Support examples for size

fishrockz, 2022-02-06:
I was reading https://docs.rust-embedded.org/book/unsorted/speed-vs-size.html and saw `cargo size --bin app -- -A` and thought i would run it on a example in one of my projects.

when i run `cargo size --example icm_filter -- -A`  or `cargo size --example icm_filter --features stm32f103 --release  -- -A` i get 

```
Failed to execute tool: size
No such file or directory (os error 2)
```

while `cargo run --example icm_filter --features stm32f103 --release` does build/run.

I cant tell if its just my case that has a issue or if size does not work for examples..

therealprof, 2022-02-06:
Weird. That is supposed to work and I just tested this works fine for me. Which OS and cargo-binutils version are you using?

fishrockz, 2022-02-07:
Fedora and i just ran cargo install yesterday so i assume the latest tag. If the examples are supposed to work then i will try again and see if i can figger it out..

adamgreig, 2022-02-07:
Have you run `rustup component add llvm-tools-preview` (from the README)? Maybe that's why `size` isn't being found.

therealprof, 2022-02-07:
@adamgreig That's a very good point, I thought we had a more useful error message for that case, but probably not; maybe we should handle that case more usefully then.

fishrockz, 2022-02-07:
> Have you run `rustup component add llvm-tools-preview` (from the README)? Maybe that's why `size` isn't being found.

That was the issue, thank you!
Sorry for failing to read the manual!

adamgreig, 2022-02-07:
Great, glad to hear it! This is definitely somewhere we should have better error reporting as it's not very obvious you have to install anything else.

fishrockz, 2022-02-07:
I asked a rustup maintainer if there was a easy way to check and apparently there isnt a easy way to check..

```
<WSalmon> Kinnison: is there a easy to use api/create to detect if a rustup componet is installed? re https://github.com/rust-embedded/cargo-binutils/issues/113#issuecomment-1031431327
<earlbot> ⤷ Support examples for size · Issue #113 · rust-embedded/cargo-binutils · GitHub
<Kinnison> No there is no such API
<WSalmon> fair
<WSalmon> thanks for the info Kinnison
<Kinnison> Some people run `rustup component list --installed` and similar
<Kinnison> but that's human UI surface, not for tooling per-se
```
:(

therealprof, 2022-02-12:
Thanks for addressing this. NB: If you added a line like "Closes #113" or "Fixes #113" then this issue would have been autoclosed by GitHub as soon as the PR is merged.

closed 2022-02-12 by therealprof

fishrockz, 2022-02-12:
> Thanks for addressing this. NB: If you added a line like "Closes #113" or "Fixes #113" then this issue would have been autoclosed by GitHub as soon as the PR is merged.

Ah yes, sorry about not closing this one.