rust-embedded/cargo-binutils #120: `cargo nm` can't find the path to existing binary

wmstack, 2022-05-16:
As per the https://docs.rust-embedded.org/embedonomicon/smallest-no-std.html, I execute the following.

```bash
cargo nm -- target/thumbv7m-none-eabi/debug/deps/app-*.o | grep '[0-9]* [^N] '
/homeo/waleed/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin/llvm-nm: error: target/thumbv7m-none-eabi/debug/deps/app-2fb83299481625ce.o: No such file or directory
```

The binary is in fact there, but `llvm-nm` doesn't notice.

```bash
rustc --version
rustc 1.60.0 (7737e0b5c 2022-04-04)
```

Emilgardis, 2022-05-16:
the `nm` command sets the CWD to the parent of the executable/binary for the current crate, so try instead `cargo nm -- deps/app-2fb83299481625ce.o`

wmstack, 2022-05-17:
Now it keeps saying `could not determine the wanted artifact.` I don't know what changed.

Emilgardis, 2022-05-17:
interesting, a glob works for me on windows. it's run as `"C:\\path\\to\\target\\debug" "rust-nm" "test_cargobinutils.exe" "deps/libcfg_if-*.rlib"` (using #121 to see cwd), the glob is kept and works in `rust-nm`

I wonder what has changed in `rust-nm` that makes this not work anymore. ping @rust-embedded/resources maybe the guide needs to be updated to account for this?

jannic, 2022-08-16:
I think the issue is twofold: First, @Emilgardis is right and `target/thumbv7m-none-eabi/debug/` needs to be removed from the path as `cargo nm` enters that directory before calling `llvm-nm`.

But on the other hand, on unix, the glob is not resolved by the called command, but by the calling shell. And the shell doesn't know about that directory prefix, looks for `deps/app-*.o` from the current directory, and fails to resolve the glob.

Something like this could work (on unix) but is ugly:
```
cargo nm -- $(cd target/thumbv7m-none-eabi/debug/ && echo deps/app-*.o)
```