rust-embedded/cargo-binutils #122: please update to cargo-metadata 0.14

sylvestre, 2022-06-19:
To facilitate the packaging of cargo-binutils (needed for a more recent version of grcov) in Debian, it would be nice to upgrade to cargo_metadata 0.14
https://crates.io/crates/cargo_metadata/versions

thanks

Emilgardis, 2022-06-19:
Is there a reason for using `--locked` in the packaging?

If you don't use `--locked`, you'll get 0.14 automatically.

Feel free to make a pr for bumping the version.

We could make a simple patch release for this and the regex bump

sylvestre, 2022-06-19:
Because they aren't compatible:
```
   Compiling cargo-binutils v0.3.5 (/home/sylvestre/dev/debian/cargo-binutils)
error[E0308]: mismatched types
  --> src/lib.rs:59:39
   |
59 |         let target_name = if let Some(Component::Normal(path)) = target_path.components().next() {
   |                                       ^^^^^^^^^^^^^^^^^^^^^^^    ------------------------------- this expression has type `std::option::Option<Utf8Component<'_>>`
   |                                       |
   |                                       expected enum `Utf8Component`, found enum `Component`

error[E0308]: mismatched types
  --> src/lib.rs:85:36
   |
85 |         if let Some(path) = search(&metadata.workspace_root, ".cargo/config") {
   |                             ------ ^^^^^^^^^^^^^^^^^^^^^^^^ expected struct `Path`, found struct `Utf8PathBuf`
   |                             |
   |                             arguments to this function are incorrect
   |
   = note: expected reference `&Path`
              found reference `&Utf8PathBuf`
note: function defined here
  --> src/lib.rs:21:4
   |
21 | fn search<'p>(path: &'p Path, file: &str) -> Option<&'p Path> {
   |    ^^^^^^     --------------  ----------

For more information about this error, try `rustc --explain E0308`.

```

with
```
diff --git a/Cargo.toml b/Cargo.toml
index ee98811..fd05a8f 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -15,7 +15,7 @@ repository = "https://github.com/rust-embedded/cargo-binutils/"
 version = "0.3.5"
 
 [dependencies]
-cargo_metadata = "0.11"
+cargo_metadata = "0.14"
 clap = "2.33"
 regex = "1.5.5"
 rustc-cfg = "0.4"

```

Emilgardis, 2022-06-19:
Ah yes of course, my bad! I'll submit a patch for this :)

sylvestre, 2022-06-19:
No worries & thanks :)

closed 2022-06-19 by bors[bot]

Emilgardis, 2022-06-20:
this change has been released now, hope it helps :)

commit 9f0a6ff7 (2022-06-19): Merge #123

123: Prepare v0.3.6 & bump deps r=burrbull a=Emilgardis

Bump deps and prepare for release

Fixes #122
PR #123 (2022-06-19): Prepare v0.3.6 & bump deps
Bump deps and prepare for release

Fixes #122