rust-embedded/cargo-binutils #127: -Z unstable flags not passed to build command

teamplayer3, 2022-08-15:
executed command:
```
cargo size --release --verbose -Z unstable-options
```
verbose output:
```
"/usr/local/rustup/toolchains/nightly-2022-06-15-x86_64-unknown-linux-gnu/bin/cargo" "build" "--release" "--message-format=json"
error: the `--out-dir` flag is unstable, pass `-Z unstable-options` to enable it
See https://github.com/rust-lang/cargo/issues/6790 for more information about the `--out-dir` flag.
error: Failed to parse crate metadata
```
binutils version:
```
0.3.6
```

kleinesfilmroellchen, 2022-11-22:
No `-Z` options are passed to the build command, e.g. `-Z build-std` which is required for embedded targets like riscv32imc-unknown-none-elf.

EnzoAlbornoz, 2023-01-09:
Try use `.cargo/config.toml` to pass these options
E.g:
```toml
# <package>/.cargo/config.toml
[unstable]
build-std = ["core", "alloc"]
```

commit 5d23ca8c (2022-08-15): Merge #128

128: correctly take unstable-features r=therealprof a=Emilgardis

fixes #127
commit 4c87c501 (2022-08-15): Merge #128

128: correctly take unstable-features r=therealprof a=Emilgardis

fixes #127
commit c5a7aad2 (2023-01-23): Merge #128

128: correctly take unstable-features r=adamgreig a=Emilgardis

fixes #127
PR #128 (2023-01-23): correctly take unstable-features
fixes #127
closed 2023-01-23 by bors[bot]