rust-embedded/cargo-binutils #135: Installing on windows with gnu fails at random times with an error about a missing build script

DJ-Dav, 2023-03-01:
Running the command `cargo install -f cargo-binutils`
gives the following error message:
error: failed to run custom build command for `quote v1.0.23`

Caused by:
  could not execute process "\AppData\Local\Temp\cargo-installurqU3j\release\build\quote-0cc442527a6b4d49\build-script-build" (never executed)

Caused by:
  The system cannot find the file specified. (os error 2)
warning: build failed, waiting for other jobs to finish...
error: failed to compile "cargo-binutils v0.3.6", intermediate artifacts can be found at...

It successfully compiles a few crates before failing, the number of successful crates compiling varies, but the error is always the same.

Emilgardis, 2023-03-01:
This error suggests something wrong with your toolchain, but it's as you suggest not a problem from `cargo-binutils` but with cargo.

I'd recommend opening an issue on https://github.com/rust-lang/cargo or chiming in on https://github.com/rust-lang/cargo/issues/9902 which seems related (and with possible solutions)

this is not something we can control

toothbrush7777777, 2023-03-10:
Sounds like it could be something to do with your anti-virus software to me. In particular, when I used McAfee in the past, I used to have similar problems.