rust-embedded/cargo-binutils #138: Consider binary releases?

thejpster, 2023-10-08:
`cargo-dist` now does the hard work of setting up binary releases, meaning people can `cargo binstall <package>` and save themselves a bunch of CI build time.

Would that work here? Or do we need to compile against the specific version of cargo someone has?

therealprof, 2023-10-08:
`cargo-binutils` only provides a nice frontend to find and call the `llvm-tools` binutils which are installed as a separate component to the rust compiler. It's (pretty much, unless the some critical CLI parameters of those tools change again)  independent of the actually installed llvm-tools and the rust compiler version. The same is true for `cargo`, all the cargo "plugins" are independent of the actual cargo version.

So it should be fine to use/provide binaries.

thejpster, 2023-10-08:
How do you find llvm-objcopy? It's not in $PATH, nor `.cargo/bin`?

Actually you can `cargo binstall cargo-binutils` because this project has been built by quickinstall. But it would be better if it was built (and signed) here instead of by someone else.