rust-embedded/cargo-binutils #139: cargo-size (any maybe others?) seem to not pass along -Z flags appropriately

jamesmunns, 2023-10-25:
I noticed that passing -Z flags seems not to work as I expect:

```
cargo build --release -Z build-std=core -Z build-std-features=panic_immediate_abort --target thumbv7em-none-eabi && arm-none-eabi-size target/thumbv7em-none-eabi/release/redacted
    Finished release [optimized + debuginfo] target(s) in 0.13s
   text     data      bss      dec      hex  filename
  15448       56   205248   220752    35e50  target/thumbv7em-none-eabi/release/redacted

cargo size --release -Z build-std=core -Z build-std-features=panic_immediate_abort --target thumbv7em-none-eabi
    Finished release [optimized + debuginfo] target(s) in 0.10s
   text     data      bss      dec      hex  filename
  19688       56   205248   224992    36ee0  redacted
```

In the second invocation, the size SHOULD be the same, but 19K is what I see when I don't enable build-std(-features).

Emilgardis, 2023-10-26:
This should be fixed on the master branch with #128, we've just not had a release since then. Not sure why

Coekjan, 2023-11-20:
I also met this problem today.