rust-embedded/cargo-binutils #141: Could not determine the wanted artifact

HaoboGu, 2024-01-18:
When I use `cargo size` I got this error: Could not determine the wanted artifact.

My project is a workspace, organized like:

```
- root
  - src
    - Cargo.toml
  - examples
    - example1
      - Cargo.toml
    - example2
      - Cargo.toml  
  - Cargo.toml
```

The example1 and example2 has different compile target, both use `src`. So I enabled `cargo-features = ["per-package-target"]` and `forced-target = "thumbv6m-none-eabi"` in Cargo.toml. 

`cargo size` worked well in the previous Rust nightly version(2023-12-31), but recently I updated Rust to 1.77-nightly(2024-01-17), `cargo size` was broken then.

burrbull, 2024-01-18:
What `Cargo.toml` inside `src`? Looks too strange.

Emilgardis, 2024-01-18:
Can you post the output of `cargo metadata --no-deps` and the also the last few lines of `cargo build --message-format=json`

This is most definitely an issue with the unstable features you're using in cargo

HaoboGu, 2024-01-19:
> What `Cargo.toml` inside `src`? Looks too strange.

`Cargo.toml` inside `src` folder is nothing special, just like any normal library's cargo config

HaoboGu, 2024-01-19:
> Can you post the output of `cargo metadata --no-deps` and also the last few lines of `cargo build --message-format=json`
> 
> This is most definitely an issue with the unstable features you're using in cargo

Output of `cargo metadata --no-deps`:

```
warning: please specify `--format-version` flag explicitly to avoid compatibility problems
{"packages":[{"name":"rmk","version":"0.1.0","id":"path+file:///Users/haobogu/Projects/keyboard/rmk/rmk#0.1.0","license":"MIT OR Apache-2.0","license_file":null,"description":"Keyboard firmware written in Rust","source":null,"dependencies":[{"name":"byteorder","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^1.4","kind":null,"rename":null,"optional":false,"uses_default_features":false,"features":[],"target":null,"registry":null},{"name":"embassy-sync","source":"git+https://github.com/embassy-rs/embassy","req":"^0.5.0","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"embassy-time","source":"git+https://github.com/embassy-rs/embassy","req":"^0.3","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"embassy-usb","source":"git+https://github.com/embassy-rs/embassy","req":"^0.1.0","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":["usbd-hid","max-interface-count-8","max-handler-count-8"],"target":null,"registry":null},{"name":"embedded-alloc","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.5","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"embedded-hal","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^1.0.0","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"embedded-storage","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.3","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"log","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.4","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"num_enum","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.7","kind":null,"rename":null,"optional":false,"uses_default_features":false,"features":[],"target":null,"registry":null},{"name":"packed_struct","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.10","kind":null,"rename":null,"optional":false,"uses_default_features":false,"features":[],"target":null,"registry":null},{"name":"static_cell","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^2","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"usbd-hid","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.6.1","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null}],"targets":[{"kind":["lib"],"crate_types":["lib"],"name":"rmk","src_path":"/Users/haobogu/Projects/keyboard/rmk/rmk/src/lib.rs","edition":"2021","doc":true,"doctest":true,"test":true}],"features":{"col2row":[],"default":["col2row"],"eeprom":[]},"manifest_path":"/Users/haobogu/Projects/keyboard/rmk/rmk/Cargo.toml","metadata":null,"publish":null,"authors":["Haobo Gu <haobogu@outlook.com>"],"categories":[],"keywords":[],"readme":"../README.md","repository":"https://github.com/haobogu/rmk","homepage":"https://github.com/haobogu/rmk","documentation":null,"edition":"2021","links":null,"default_run":null,"rust_version":null},{"name":"rmk-rp2040","version":"0.1.0","id":"path+file:///Users/haobogu/Projects/keyboard/rmk/boards/rp2040#rmk-rp2040@0.1.0","license":"MIT OR Apache-2.0","license_file":null,"description":"Keyboard firmware written in Rust","source":null,"dependencies":[{"name":"cortex-m","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.7.7","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":["inline-asm"],"target":null,"registry":null},{"name":"cortex-m-rt","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.7.3","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"defmt","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.3","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"defmt-rtt","source":"registry+https://github.com/rust-lang/crates.io-index","req":"^0.4","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"embassy-executor","source":"git+https://github.com/embassy-rs/embassy","req":"^0.5.0","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":["defmt","arch-cortex-m","executor-thread","integrated-timers"],"target":null,"registry":null},{"name":"embassy-futures","source":"git+https://github.com/embassy-rs/embassy","req":"^0.1.0","kind":null,"rename":null,"optional":false,"uses_default_features":true,"features":[],"target":null,"registry":null},{"name":"embassy-rp","source":"git+https://g [... cut ...] buginfo":2,"debug_assertions":true,"overflow_checks":true,"test":false},"features":["col2row","default","eeprom"],"filenames":["/Users/haobogu/Projects/keyboard/rmk/target/debug/librmk.rlib","/Users/haobogu/Projects/keyboard/rmk/target/debug/deps/librmk-55f497bcf3825df7.rmeta"],"executable":null,"fresh":true}
{"reason":"build-finished","success":true}
    Finished dev [optimized + debuginfo] target(s) in 0.08s
```

BTW, `cargo build` works well, `cargo size` just cannot print out the binary info

HaoboGu, 2024-01-19:
A few more information that I could provide:

The workspace's `Cargo.toml`:
```toml
[workspace]
resolver = "2"
members = ["rmk", "boards/*"]
default-members = ["boards/stm32h7", "rmk"]
```

`Cargo.toml` of example, which placed at the `board/stm32h7` folder(aka example/example1 in previous toy demo):

```toml
cargo-features = ["per-package-target"]

[package]
name = "rmk-stm32h7"
version = "0.1.0"
authors = ["Haobo Gu <haobogu@outlook.com>"]
description = "Keyboard firmware written in Rust"
homepage = "https://github.com/haobogu/rmk"
repository = "https://github.com/haobogu/rmk"
readme = "../../README.md"
edition = "2021"
license = "MIT OR Apache-2.0"

forced-target = "thumbv7em-none-eabihf"
[dependencies]
...
[[bin]]
name = "rmk-stm32h7"
test = false
bench = false

```

HaoboGu, 2024-01-19:
@Emilgardis @burrbull it seems `cargo size` failed to parse the output of `cargo build`?

Emilgardis, 2024-01-19:
The problem is due to a mismatch on how `cargo metadata` and `cargo build --message-format=json` represents package ids.

This seems to be a recent change, I'll see if I can locate it. A fix should be done though, parsing package ids from cargo is in my experience notoriously hard to do properly

Emilgardis, 2024-01-19:
```
workspace members package_id : [PackageId { repr: "path+file:///Users/haobogu/Projects/keyboard/rmk/rmk#0.1.0" }, PackageId { repr: "path+file:///Users/haobogu/Projects/keyboard/rmk/boards/rp2040#rmk-rp2040@0.1.0" }, PackageId { repr: "path+file:///Users/haobogu/Projects/keyboard/rmk/boards/stm32h7#rmk-stm32h7@0.1.0" }]       
artifact.package_id: PackageId { repr: "embedded-storage 0.3.1 (registry+https://github.com/rust-lang/crates.io-index)" }
artifact.package_id: PackageId { repr: "rmk-stm32h7 0.1.0 (path+file:///Users/haobogu/Projects/keyboard/rmk/boards/stm32h7)" }
artifact.package_id: PackageId { repr: "rmk 0.1.0 (path+file:///Users/haobogu/Projects/keyboard/rmk/rmk)" }
```

We compare the package ids here: https://github.com/rust-embedded/cargo-binutils/blob/5c38490e1abf91af51d0a345bb581e37facd28ff/src/lib.rs#L424

and it fails because `path+file:///Users/haobogu/Projects/keyboard/rmk/rmk#0.1.0` != `rmk 0.1.0 (path+file:///Users/haobogu/Projects/keyboard/rmk/rmk)`

Emilgardis, 2024-01-19:
Aha!

https://github.com/rust-lang/cargo/pull/13311#issuecomment-1899507336

So, update nightly to atleast `cargo 1.77.0-nightly (1ae631085 2024-01-17)` with `rustup update nightly` and it should work again. That version of cargo was released first today.

Emilgardis, 2024-01-19:
Also, you may want to specify `--lib` or `--bin <name>` to disambiguate

closed 2024-01-19 by Emilgardis

HaoboGu, 2024-01-20:
okay! I'll try it later😉