rust-embedded/cargo-binutils #153: All cargo-binutils commands fail with errors.

cnwzhu, 2024-12-05:
```
PS C:\Users\wang-\IdeaProjects\testbl> cargo -V
cargo 1.85.0-nightly (4c39aaff6 2024-11-25)
PS C:\Users\wang-\IdeaProjects\testbl> cargo nm -V                                   
cargo-nm 0.3.6

PS C:\Users\wang-\IdeaProjects\testbl> cargo nm --release --target thumbv6m-none-eabi
    Finished `release` profile [optimized] target(s) in 0.00s
error: Could not determine the wanted artifact
PS C:\Users\wang-\IdeaProjects\testbl> cargo nm --release
    Finished `release` profile [optimized] target(s) in 0.00s
error: Could not determine the wanted artifact
PS C:\Users\wang-\IdeaProjects\testbl> cargo objcopy --release --target thumbv6m-none-eabi -- -O binary app.bin 
    Finished `release` profile [optimized] target(s) in 0.01s
error: Could not determine the wanted artifact
```

Emilgardis, 2024-12-05:
Can you try adding `--bin <bin_name>`

cnwzhu, 2024-12-05:
> Can you try adding `--bin <bin_name>`

This project is a staticlib 🙂‍↔️

Emilgardis, 2024-12-05:
Aha, then try `--lib`

cnwzhu, 2024-12-05:
> Aha, then try `--lib`

It work. Thanks.

Emilgardis, 2024-12-05:
We should improve the error message to suggest these flags