rust-embedded/cargo-binutils #161: panic with the published v0.4.0

zh-jq-b, 2025-08-27:
When run
```rust
cargo install --force cargo-binutils@0.4.0
cargo profdata -- --help
```
It will panic with the following message:
```text
thread 'main' panicked at /home/zhangjingqiang/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/clap_builder-4.5.46/src/parser/matches/arg_matches.rs:185:17:
arg `no-default-features`'s `ArgAction` should be one of `SetTrue`, `SetFalse` which should provide a default
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Emilgardis, 2025-09-26:
This is now fixed on the main branch. `cargo install cargo-binutils --git https://github.com/rust-embedded/cargo-binutils`

closed 2025-09-26 by Emilgardis

Sytten, 2026-01-03:
Can this be released please, facing this issue